openhwfoundation/cve2 #37: Update Makefile(s) to support Cadence Xcelium and other TBD simulators for the above

DBees, 2023-01-24:


MikeOpenHWGroup, 2023-09-12:
[CORE-V-VERIF Pull-Request #2195](https://github.com/openhwgroup/core-v-verif/pull/2195) adds support for VCS.

MikeOpenHWGroup, 2023-10-05:
As far as I can tell, the Makefiles for CV32E20 have supported Questasim (VSIM) as of [CORE-V-VERIF Pull-Request #2003](https://github.com/openhwgroup/core-v-verif/pull/2003).  In any case, I can report that a `ci_check` regression runs cleanly with both VSIM and VCS on the current head of the `cv32e20/dev` branch ([2f551d3](https://github.com/openhwgroup/core-v-verif/commit/2f551d3a00d3d57d60e71bcb6de8171a5a775c6a)).

As such, I think this Task can be closed.

MarioOpenHWGroup, 2024-03-11:
In order to complete this task, the merge of `cv32e20/dev` to `master` is required. We need to exclude the `cv32e20` folder in the merge as it is currently in a different repo

Floadv298, 2024-03-25:
@MarioOpenHWGroup that means change to 'In Progress'?