openhwfoundation/cve2 #58: JEDEC manufacturer ID for CV32E20

MikeOpenHWGroup, 2023-02-14:
We do not yet have a JEDEC Machine Architecture ID assigned to the CV32E20.  This will be required to populate the `mvendorid` CSR.  See the CV32E40P CSR spec for an example ([link](https://docs.openhwgroup.org/projects/cv32e40p-user-manual/en/latest/control_status_registers.html#machine-vendor-id-mvendorid)).

I'm labelling this as a "bug" since it will require a change to the RTL.

christian-herber-nxp, 2023-02-14:
@MikeOpenHWGroup: could you change mvendorid to march, which is the CSR in question

MikeOpenHWGroup, 2023-02-14:
Ah.  Well AFAIK, `marchid` is completely up to us and creation and allocation of these is in @davideschiavone's wheelhouse.

christian-herber-nxp, 2023-02-15:
agree. however, it seems that up to date, OpenHW group has allocated a marchid in the global registry of RVI https://github.com/riscv/riscv-isa-manual/blob/master/marchid.md, which seems unnecessary considering OpenHW Group does have a JEDEC vendor ID.

davideschiavone, 2023-02-15:
let's move this discussion on MM on the Core TG channel

closed 2023-09-26 by christian-herber-nxp