openhwfoundation/cve2 #62: Verification environment integration

LeeHoff, 2023-02-17:


MikeOpenHWGroup, 2023-02-18:
Hi @LeeHoff, what this the goal of this task?

MikeOpenHWGroup, 2023-09-12:
Given that the CV32E20 has been integrated into the CORE-V-VERIF environment and that ci_check regressions are running under multiple simulators, I would suggest that this task is complete.  There is more work to do, most importantly integrating ImperasDV, but this is tracked under task #32.

MikeOpenHWGroup, 2024-03-11:
Re-assigning to Mario and myself.   The completion criteria for this Task:
- Integration of CV32E20 into the [cv32e20-dv](https://github.com/openhwgroup/cv32e20-dv) (CORE-V-VERIF) environment.
- `ci_check` regressions are running under multiple simulators.
- Integration with the Spike reference model.
- Conclusion of #32.
- Merging of all cve2 related updates to the master branch of CORE-V-VERF.