openhwfoundation/cve2 #88: Implement directed tests to exercise all debug features

LeeHoff, 2023-02-17:


MikeOpenHWGroup, 2023-09-12:
[CORE-V-VERIF Pull-Request #2128](https://github.com/openhwgroup/core-v-verif/pull/2128) gets the `debug_test` running in on the CV32E20 in CORE-V-VERIF.

- Stand-up the "Debug Control" virtual peripheral will be complete once the above PR is merged in.
- Review of custom debug_test is on-going.  14 of 20 sub-tests are passing.

MikeOpenHWGroup, 2023-10-05:
A partial version of the [debug_test](https://github.com/openhwgroup/core-v-verif/blob/2f551d3a00d3d57d60e71bcb6de8171a5a775c6a/cv32e20/tests/programs/custom/debug_test/debug_test.c#L1-L2) has been merged into the CV32E20 environment and is passing.  [CORE-V-VERIF Pull-Request #2210](https://github.com/openhwgroup/core-v-verif/pull/2221) exists to add this test to the `ci_check` regression.

Work to add remaining "sub-tests" implemented within [debug_test](https://github.com/openhwgroup/core-v-verif/blob/2f551d3a00d3d57d60e71bcb6de8171a5a775c6a/cv32e20/tests/programs/custom/debug_test/debug_test.c#L1-L2) is on-going.