openhwfoundation/cve2 #110: Inconsistent reset address

NignetShark, 2023-04-03:
### Bug Description

I noticed that reset address was moved from `boot_addr + 0x80` to `boot_addr + 0x0` but some files are still using the old address, such as :
* `sw/simple_system/common/ctr0.S`,  line 100 :
```s
// reset vector
.org 0x80
jal x0, reset_handler
```
* `sw/simple_system/common/link.ld`,  line 22 :
```ld
_entry_point = _vectors_start + 0x80;
```
I don't know if it's a legacy from the Ibex documentation but `doc/03_reference/exception_interrupts.rst` and `doc/02_user/integration.rst` still refer to the old reset address which might be misleading.

christian-herber-nxp, 2023-06-02:
Decision is that reset address will be 0. this should be fixed in the files pointed out.

szbieg, 2023-06-07:
I have no way to test the requested modification of the crt0.S as the current test environment is using bare-metal-programs using the bsp/crt0.S.