openhwfoundation/cve2 #135: Parameter overrides are not allowed for localparams

MikeOpenHWGroup, 2023-07-28:
### Bug Description

The top-level module [cve2_top_tracing.sv](https://github.com/openhwgroup/cve2/blob/b94bca939e7cc305a288e290dae084e2bff89d0c/rtl/cve2_top_tracing.sv#L1) attempts to override three localparam in [cve2_top.sv](https://github.com/openhwgroup/cve2/blob/b94bca939e7cc305a288e290dae084e2bff89d0c/rtl/cve2_top.sv#L1):
1. `RV32B` on [line 123](https://github.com/openhwgroup/cve2/blob/b94bca939e7cc305a288e290dae084e2bff89d0c/rtl/cve2_top_tracing.sv#L123).
2. `DbgTriggerEn` on [line 124](https://github.com/openhwgroup/cve2/blob/b94bca939e7cc305a288e290dae084e2bff89d0c/rtl/cve2_top_tracing.sv#L124).
3. `DbgHwBreakNum` on [line 125](https://github.com/openhwgroup/cve2/blob/b94bca939e7cc305a288e290dae084e2bff89d0c/rtl/cve2_top_tracing.sv#L125).

This prevents the RTL model from compiling when `cve2_top_tracing.sv` is compiled as the top-level (as it is in the CORE-V-VERIF environment).

LeeHoff, 2023-07-28:
this is fixed in my PR #132.  Previously the cve2_top was instanced in core-v-verif/cv32e20/tb/uvmt/uvmt_cv32e20_dut_wrap, now when we instance cve2_top_tracing it needs the same parameters and I/O as cve2_top.  This is the change  in PR #132.

MikeOpenHWGroup, 2023-07-28:
Right you are @LeeHoff.  In fact, that is exactly what you told me in a MM post.  Mon bad.

#132 is now merged in an these errors have disappeared.

closed 2023-07-28 by MikeOpenHWGroup

LeeHoff, 2023-07-28:
@MikeOpenHWGroup It's good to have a bug created to say why things changed.