openhwfoundation/cve2 #142: PoR value of mcountinhibit CSR is not documented

MikeOpenHWGroup, 2023-08-29:
### Bug Description

By visual inspection of the RTL, the power-on-reset value of `mcountinhibit` is 'h0.  I believe this to be a valid state of affairs, but this is different that other CORE-V cores and is not documented in the User Manual.  Check out the CV32E40P User Manual's [CSR Chapter](https://cv32e40p.readthedocs.io/en/latest/control_status_registers.html#machine-counter-inhibit-register-mcountinhibit) for an example.

MaartenArts, 2023-09-01:
The performance counter reset values are documented here main/doc/03_reference/performance_counters.rst. There is a difference in these reset values between the cv32e40 and cv32e20 cores. It might be good to have consistency for these reset values between the cv32 cores. Also for low power consumption the performance counters are expected to be disabled after reset.

MaartenArts, 2023-09-01:
There is also a difference in reset values between cv32e20 and cv32e40 for the mhpmevent csr registers.