openhwfoundation/cve2 #164: [BUG] Incorrect instruction counting (MINSTRET) when executing a DRET in Machine mode.

LeeHoff, 2023-10-16:
### Bug Description

On the execution of dret when in machine mode, the DUT is incrementing minstret.  As the dret causes an exception it is not retired and as such should not increment minstret.  Found during simulation of the debug_test with step-and-compare.

Privileged Specification, 3.1.10 Hardware Performance Monitor: The minstret CSR counts the number of instructions the hart has retired

szbieg, 2023-11-03:
Related bug: https://github.com/lowRISC/ibex/issues/1132

MikeOpenHWGroup, 2023-12-01:
According the `Unprivileged ISA, Document Version 20191213`, in Section 3.3.1:
```
ECALL and EBREAK cause synchronous exceptions, they are not considered to retire, and should not increment the minstret CSR.
```
Although this is not explicitly stated anywhere else, the above implies that _no_ synchronous exceptions are considered to retire.  Assuming that interpretation is correct and knowing that the execution of `dret` when not in debug-mode is a synchronous exception, I agree that we should not be incrementing `minstret` in this case.

LeeHoff, 2023-12-01:
To duplicate this error, make this one line change to the cv32e20/tb/uvmt/uvmt_cv32e20_imperas_dv_wrap.sv file:
line 587 was : 
    void'(rvviRefCsrSetVolatile(hart_id, `CSR_MINSTRET_ADDR     ));
change it to:
//    void'(rvviRefCsrSetVolatile(hart_id, `CSR_MINSTRET_ADDR     ));

Then run the debug_test with USE_ISS=YES.  This is the command line I use with some extra logging that Duncan G. from Imperas suggested:

make test TEST=debug_test USE_ISS=YES WAVES=YES USER_RUN_FLAGS="+cve2_tracer_enable=1 +IDV_TRACE2LOG=1 +UVM_MAX_QUIT_COUNT=10,NO"

The first error I get has this output:
Info 89246: 'refRoot/cpu', 0x000000000000078a(main+606): Machine 7b200073 dret
Warning (RISCV_ILL) CPU 'refRoot/cpu' 0x0000078a 7b200073 dret: Illegal instruction - not debug mode
Info   mstatus 00000080 -> 00001800 [SD:0 TW:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0->3 VS:0(Off) MPIE:1->0 UBE:0 MIE:0]
Info   mepc 00000742 -> 0000078a
Info   mtval 7b0027f3 -> 7b200073
Info 89247: 'refRoot/cpu', 0x0000000000000100(vector_table): Machine 4170006f j       d16
Info (IDV) Instruction executed prior to mismatch '0x100(vector_table+0): 4170006f j       d16'
Error (IDV) CSR register value mismatch (HartId:0, PC:0x00000100 vector_table+0):
Error (IDV) Mismatch 0> CSR b02 (minstret)
Error (IDV)   . dut:0x00015c83
Error (IDV)   . ref:0x00015c82
UVM_ERROR @ 1689001.800 ns : idvPkg.sv(59) reporter [] uvmt_cv32e20_tb.imperas_dv.trace2api.state_compare @ 1689002.000 ns: MISMATCH

PR #180 (2024-02-26): Do not count dret instruction when not in debug mode - it won't retire
fixes #164
commit 00bad970 (2024-02-26): Do not count dret instruction when not in debug mode - it won't retire (#180)

fixes #164
closed 2024-02-26 by davideschiavone