openhwfoundation/cve2 #169: Clarify core configuration parameters of the "golden configurations"

MikeOpenHWGroup, 2023-11-08:
### Bug Description

The User Manual identifies four [golden configurations](https://docs.openhwgroup.org/projects/cve2-user-manual/en/latest/01_specification/index.html#golden-configurations), but does not explicitly state the parameterization required to specify these configurations, nor does it correlate any of these configurations with the terms "micro" or "small" used in the original [Project Concept slides](https://github.com/openhwgroup/programs/blob/master/Project-Descriptions-and-Plans/CVE20/OpenHW-CV32E20-PC-Cores-TG-Meeting_210628.ppt).

MikeOpenHWGroup, 2023-11-21:
Just for the record, here is a summary of my comments on [MatterMost](https://mattermost.openhwgroup.org/all-users/pl/9usb1qmkrbg65nqg6imyyqrn4h) before this issue was generated.  I am repeating them here since not everyone who has access to this GitHub repo will automatically have access to OpenHW MatterMost channels.

- The original [Project Concept slides](https://github.com/openhwgroup/programs/blob/master/Project-Descriptions-and-Plans/CVE20/OpenHW-CV32E20-PC-Cores-TG-Meeting_210628.ppt) indicate that the "micro" and "small" configurations would be verified, and that the "small" configuration supports a 3 cycle multipler.
- The User Manual identifies four [golden configurations](https://docs.openhwgroup.org/projects/cve2-user-manual/en/latest/01_specification/index.html#golden-configurations), but does not explicitly state the parameterization required to specify these configurations, nor does it correlate any of these configurations with the terms "micro" or "small".
- The [Parameters](https://docs.openhwgroup.org/projects/cve2-user-manual/en/latest/02_user/integration.html#parameters) section in the `Core Integration` chapter of the User Manual implies that`RV32MFast` specifies a "3-4 cycle multiplier, iterative divider" math unit.
- The [Project Launch](https://github.com/openhwgroup/programs/blob/master/Project-Descriptions-and-Plans/CVE20/Project-Launch-for-CV32E20-220209.md) is similarily vague.

christian-herber-nxp, 2023-11-22:
Related:
https://github.com/openhwgroup/cve2/issues/57
https://github.com/openhwgroup/cve2/issues/146