openhwfoundation/cve2 #175: [BUG] Specification either not consistent or contradictory

szbieg, 2023-11-30:
### Bug Description

Specification item [PVL-30](https://github.com/openhwgroup/cve2/blob/main/doc/01_specification/index.rst?plain=1#L254) contradicts with [MEM-70](https://github.com/openhwgroup/cve2/blob/main/doc/01_specification/index.rst?plain=1#L663)

christian-herber-nxp, 2023-11-30:
please elaborate why it is contradicting. one requirement is regarding the core interface, one regarding the coreplex interface

szbieg, 2023-11-30:
These two requirements has to be considered together as a common CVE32E20 specification, as they are tightly related to each other. They do not exist in their own vacuum, on the contrary: forcing one encoding different from the other is obviously resulting in additional logic that would not be needed if the implementation was not (explicitly) specified, but rather requested functionality.

I see a solution as to either:

- relax encoding specification of the core interface (not explicit encoding), or

- harmonize both interfaces encodings.