openhwfoundation/cve2 #198: Rename IBEX to CVE2

DBees, 2024-04-10:


MikeOpenHWGroup, 2025-02-13:
Hi @cairo-caplan, I believe this task is completed, but as @davide is on vacation, please review the status of the sources.  If there are any outstanding filenames or module names to update, please do so.  Otherwise, we can mark this task as complete.

cairo-caplan, 2025-02-13:
All items of the task are done, most of it since #25 . 

However, it is all merged onto a `main` branch, instead of a `master`, as it is the only existing branch.

MikeOpenHWGroup, 2025-02-13:
> All items of the task are done

Excellent.  Please click on the sub-tasks in the Description of this task.

> However, it is all merged onto a main branch, instead of a master, as it is the only existing branch.

That is expected.  Several years ago GitHub changed the name of the "trunk"  from `master` to `main`.  Many of our repos were created before this change so they have master branches.  This repo was created after the change, so it has a main branch.  They has the same purpose, just different names.

I have updated the Description to say `main`.

closed 2025-03-03 by MikeOpenHWGroup