openhwfoundation/cve2 #200: Values for MVENDORID, MARCHID and MIMPID

DBees, 2024-04-10:


DBees, 2024-04-25:
Establishing the numbers:
VendorID is from JEDEC - we have this
ArchID is assigned from RISC-V and indicates the processor - not done yet
ImpID is 0 as it is the first implementation of this architecture

Documentation not updated

MikeOpenHWGroup, 2025-01-10:
I believe these are the correct values:

| CORE-V Name | mvendorid  | marchid    | mimpid     |
| -------------------- | --------------- | -------------- | ------------- |
| CV32E20         | 0x00000602 | 0x00000023 | 0x00000000 |

As Duncan indicated, `mvendorid` has been assigned to OpenHW by [JEDEC](https://www.jedec.org/).  The `marchid` has been assigned to the the CV32E20 by [RISC-V International](https://github.com/riscv/riscv-isa-manual/blob/main/marchid.md).  Lastly, by convention, the value of `mimpid` for the first formal release of any CORE-V core is 0x0.

eladawy88, 2025-01-17:
1. I check the RTL implementation of values of mvendorid, marchid and mimpid.

-  For value of  "mvendorid", The assigned value is "0x00000602" as shown in the following line 
https://github.com/openhwgroup/cve2/blob/4b782ac0d7f2d3166a26cd04a807697ea536aeff/rtl/cve2_pkg.sv#L563

-  For value of  "marchid", The assigned value is "0x00000023" is equivalent to "0d35"  as shown in the following line
https://github.com/openhwgroup/cve2/blob/4b782ac0d7f2d3166a26cd04a807697ea536aeff/rtl/cve2_pkg.sv#L564
https://github.com/openhwgroup/cve2/blob/4b782ac0d7f2d3166a26cd04a807697ea536aeff/rtl/cve2_pkg.sv#L561

-  For value of  "mimpid", The assigned value is "0x00000000" is  as shown in the following line
https://github.com/openhwgroup/cve2/blob/4b782ac0d7f2d3166a26cd04a807697ea536aeff/rtl/cve2_pkg.sv#L570

2. I run the hello-world program test, the program return the correct Expected values for mvendorid, marchid and mimpid defined in the following lines 
https://github.com/openhwgroup/cv32e20-dv/blob/29e5f2e5e1bf6a4662149884f0e412224d5b9258/tests/programs/custom/hello-world/hello-world.c#L31

https://github.com/openhwgroup/cv32e20-dv/blob/29e5f2e5e1bf6a4662149884f0e412224d5b9258/tests/programs/custom/hello-world/hello-world.c#L33

https://github.com/openhwgroup/cv32e20-dv/blob/29e5f2e5e1bf6a4662149884f0e412224d5b9258/tests/programs/custom/hello-world/hello-world.c#L34

eladawy88, 2025-02-01:
document is updated and waiting for reviews in the following pull request 
https://github.com/openhwgroup/cve2/pull/278

MikeOpenHWGroup, 2025-02-03:
Update: Mohammed to update the UM as per @davideschiavone feedback.

cairo-caplan, 2025-03-28:
This task has the same name as #12.
I suggest changing its title to "Values for MVENDORID, MARCHID and MIMPID in the User Manual" for clarification

MikeOpenHWGroup, 2025-10-06:
We _think_ this task is complete.   Assigning to @cairo-caplan to double-check the implementation (RTL) and documentation.