openhwfoundation/cve2 #219: Merge all updates for CV32E20 on CORE-V-VERIF to master branch

DBees, 2024-04-10:


MikeOpenHWGroup, 2025-03-19:
Progress to date:
- [cv32e20-dv](https://github.com/openhwgroup/cv32e20-dv) repo has been created
- `dev` branch is working as main development branch.

Recommend we close this issue when:
- [ ] we complete the next merge from `dev` to `main`, and
- [ ] `cv32e20-dv/main` is "pointing" to the `openhwgroup` repo of core-v-verif (see [ExternalRepos.mk](https://github.com/openhwgroup/cv32e20-dv/blob/c3c2d4b055b33e2046d74e1c56cb18f2c5e5c32f/sim/ExternalRepos.mk#L17)).

MikeOpenHWGroup, 2025-12-09:
The [openhwgroup/cv32e20-dv](https://github.com/openhwgroup/cv32e20-dv) repo is up and running.  For now, it makes sense to continue working on the `dev` branch, so I am waiving the "complete the merge from `dev` to `main` from the above completion checklist for this task.

[ExternalRepos.mk](https://github.com/openhwgroup/cv32e20-dv/blob/c2fcaf92c18319d5f4ee9ec7fbd7ac894963197d/sim/ExternalRepos.mk#L17) now points to he openhwgroup repo of core-v-verif (`cv32e20-dv/dev` branch).