openhwfoundation/cve2 #221: Create stand-alone RISCOF environment for the CV32E20

DBees, 2024-04-11:


MikeOpenHWGroup, 2025-01-20:
Note that all other CORE-V cores have opted _not_ to do this, in favour of integrating the RISCOF tests directly into the UVM environment.  This will also be done for the CV32E20 in task #223.   Nevertheless, it is worthwhile to do this as in stand-alone testbench as an aid to integration and debugging.

MikeOpenHWGroup, 2025-01-20:
Start by reviewing the [RISC-V Compatibility Framework](https://riscof.readthedocs.io/en/stable/intro.html).  Next, have a look at the [CV32E40P-RISCOF README](https://github.com/MikeOpenHWGroup/cv32e40p-riscof/blob/main/README.md) and ensure you have access to all the prerequists.  Once these are in place, the [run script](https://github.com/MikeOpenHWGroup/cv32e40p-riscof/blob/main/run.sh) _should_ work for you.

eladawy88, 2025-02-05:
@MikeOpenHWGroup Can you please create repo for CV32E20-RISCOF project?