openhwfoundation/cve2 #232: Run RISC-V ISA “custom” tests with Spike ISS enabled

DBees, 2024-04-16:


MarioOpenHWGroup, 2024-05-02:
Almost all of the ci_checks tests running and passing with cve2 and spike working in tandem mode. 

```
CI Check results:
                 vsim-interrupt_test.log :  FAILED
           vsim-corev_rand_interrupt.log :  FAILED
           vsim-corev_rand_interrupt.log :  FAILED
    vsim-corev_rand_jump_stress_test.log :  PASSED
               vsim-csr_instructions.log :  PASSED
                        vsim-illegal.log :  PASSED
          vsim-corev_rand_instr_test.log :  PASSED
                    vsim-hello-world.log :  PASSED
vsim-corev_rand_arithmetic_base_test.log :  PASSED
  vsim-riscv_arithmetic_basic_test_0.log :  PASSED
CI Check FAILED: Expected 11 tests to run but found only 10 PASSED or FAILED messages
```

DBees, 2024-05-02:
24/5/2 - many tasks passing

MarioOpenHWGroup, 2024-05-02:
related task https://github.com/openhwgroup/cva6/issues/2094

MarioOpenHWGroup, 2024-05-17:
Currently status of the ci_check
```
CI Check results:
                     vsim-debug_test.log : ABORTED
    vsim-corev_rand_jump_stress_test.log :  PASSED
               vsim-csr_instructions.log :  PASSED
                 vsim-interrupt_test.log :  PASSED
                        vsim-illegal.log :  PASSED
          vsim-corev_rand_instr_test.log :  PASSED
                    vsim-hello-world.log :  PASSED
vsim-corev_rand_arithmetic_base_test.log :  PASSED
  vsim-riscv_arithmetic_basic_test_0.log :  PASSED
           vsim-corev_rand_interrupt.log :  PASSED
           vsim-corev_rand_interrupt.log :  PASSED
```

MarioOpenHWGroup, 2024-05-17:
Modifications to the cve2 bsp in order to align the corev-dv tests with the testbench

MikeOpenHWGroup, 2024-05-17:
Can this task now be marked complete?