openhwfoundation/cve2 #263: [BUG] build simple_system from top level Makefile fails

TiagoTeixeira-synthara, 2024-04-30:
### Bug Description

Building the `simple_system` example (from top-level Makefile `make build-simple-system`) fails due a missing dependency: `cve2_pcounts` (full name `lowrisc_dv_verilator_cve2_pcounts`).

> `simple_system` <- `simple_system_core` <- `cve2_pcounts` (simplified names)

It seems that before the rename on #25 it was called `ibex_pcounts`, located under `dv/verilator`, but had been removed with issue #21 .

From looking around, and as documentation tells, verilator _support_ was dropped in favor of core-v-verif. As a side node, perhaps it should be made clear that `cv32e20` currently has its own branch on core-v-verif (`cv32e20/dev`)?

It seems then that the `simple_system` example is _obsolete_.

Is there a way to still be able to build/run the simple_system to/with verilator? (I imagine restoring the `ibex_pcounts`/`cve2_pcounts` files, maybe directly into the simple_system directory, would suffice.)

Should the example be dropped completely in favor of core-v-verif?

Thanks

MikeOpenHWGroup, 2024-04-30:
Thanks for this issue @TiagoTeixeira-synthara.   As you probably know, the CVE2 is a fork of Ibex and we have no plans to support the `simple-system`.  If you are planning to simulate this core it is best to use CORE-V-VERIF.

I will close this issue with a "wont-fix" label and create a task to ensure that we make a conscious decision about what to do with `simple-system` (and then update the directory tree and documentation to reflect that).

closed 2024-04-30 by MikeOpenHWGroup

MariaElenaDAgostino-synthara, 2024-05-22:
Thank you @MikeOpenHWGroup, where can I find the definition of the vector table for the CVE2?

I am trying to use the same vector table of the CV32E40P, but something is not working. 

Can you point me  to the relevant documentation or a snippet of the tb code?

MikeOpenHWGroup, 2024-05-22:
Most CORE-V cores have User Manuals that are published on ReadTheDocs.  From the GitHub repo you can find a link to it under the "About" section on the right-hand pane of the webpage.  Better yet, follow [this](https://docs.openhwgroup.org/projects/cve2-user-manual/en/latest/) link.  :wink:

By the way, please do not read the RTL or TB code to figure out how any of the CORE-V cores work.  If you cannot answer your question by reading the User Manual, please create an Issue against the User Manual.  At OpenHW, we take documentation very seriously and bugs in the documentation are treated the same as bugs in the RTL.

DanieleParravicini-Synthara, 2024-05-23:
Hey @MikeOpenHWGroup, I am a collegue of @MariaElenaDAgostino-synthara 

By looking at the ReadTheDocs you pointed us to I landed here https://docs.openhwgroup.org/projects/cve2-user-manual/en/latest/02_user/integration.html.

I could find just a reference on the [boot_addr_i](https://docs.openhwgroup.org/projects/cve2-user-manual/en/latest/02_user/integration.html) but I do not find any information on the interrupt vector table.

We tried to use a vector table that was compatible with this other [version of Ibex](https://github.com/lowRISC/ibex) and used [its documentation on ReadTheDocs](https://ibex-core.readthedocs.io/en/latest/03_reference/exception_interrupts.html). But it does not work, maybe something changed with the interrupt vector table format. 

We will share some waveforms to show the problem we are facing. 

We could submit a PR to fix the simple-system firmware if we have all the information.

Kind regards,
Daniele

MikeOpenHWGroup, 2024-05-23:
Hi @DanieleParravicini-Synthara, please do not introduce a second topic to a closed Issue. We are very strict about having one-and-only-one topic per GitHub Issue.  I think this is an important topic, so if you could open a **_new_** issue that would be great.  Thanks for your understanding.

DanieleParravicini-Synthara, 2024-05-27:
refer to #268