openhwfoundation/cve2 #264: Clarify use of Ibex "simple-system"

MikeOpenHWGroup, 2024-04-30:
### Task Description

As reported in #263, this repository has an [examples/simple_system](https://github.com/openhwgroup/cve2/blob/1786fbfab7e0e2128fbf94495e9bd3c4896d526b/examples/simple_system/README.md?plain=1#L1) directory.  Typically CORE-V core repositories do not include such examples as our members tend to have their own in-house, proprietary systems.  We need to make a conscious decision about what to do about all of the "examples" in the directory of the same name.

### Description of Done

One of the following:
1. Delete [cve2/examples]() altogether and remove any references to it in all READMEs and the User Manual.
2. Update the examples to work in the CVE2 context.

MarioOpenHWGroup, 2024-05-23:
Flo, Davide, Duncan and Mario agreed to option 1. @MikeOpenHWGroup do you agree?

MikeOpenHWGroup, 2024-05-23:
Yes, I do agree.  I'll take this one.  Expect a PR in the coming days...

MikeOpenHWGroup, 2024-05-30:
As #272 is merged in, this task is Complete.

closed 2024-05-30 by MikeOpenHWGroup