openhwfoundation/cve2 #267: [BUG] <misalgined access>

shanshuixiangyi, 2024-05-21:
### Bug Description

obi2ahbm_adapter.sv..   if byte access 24bits , bridge don't split 2 trans

davideschiavone, 2024-05-21:
hi @MikeOpenHWGroup and @shanshuixiangyi - the CVE2 supports misaligned on the OBI interface, so it is very well possible that the bridge has this bug -

MikeOpenHWGroup, 2024-05-21:
Thanks @davideschiavone, I have removed the `question` label and this is now officially a bug!  :wink: 

Hi @shanshuixiangyi, do you have an example test-program to exercise this?

Jiahua-Gong, 2024-05-23:
[image]
risc-v load store uint can transmit red select in the pitcture. but bridge can't support this byte condition.  I have modified the rtl to support this condition. I am learning how to submit code and then make a fix commit using Git.

MikeOpenHWGroup, 2024-05-23:
> I am learning how to submit code and then make a fix commit using Git.

Thank you very much @Jiahua-Gong!  I notice that the CONTRIBUTING file for this repo is not complete (I will fix that).  In the meantime, please review `CONTRIBUTING.md` in [CORE-V-VERIF](https://github.com/openhwgroup/core-v-verif), especially the [Contributor Agreement](https://github.com/openhwgroup/core-v-verif/blob/master/CONTRIBUTING.md#contributor-agreement) section.